RobTillaart/FRAM_I2C #4: Error by reading internal Metadata

PraxisSoft, 2021-01-11:
Your old Function is invalid. The new Fuction should be like this:
// metadata is packed as  [....MMMM][MMMMDDDD][PPPPPPPP]
// M = manufacturerID
// D = density => memsize = 2^D KB
// P = product ID (together with D)
uint16_t FRAM::getMetaData(uint8_t field)
{
  if (field > 2) return 0;

  Wire.beginTransmission(FRAM_SLAVE_ID_);
  Wire.write(_address << 1);
  Wire.endTransmission(false);
  int x = Wire.requestFrom(FRAM_SLAVE_ID_, (uint8_t)3);
  if (x != 3) return -1;

  uint32_t value = 0;
  value = Wire.read();
  value = value << 8;     // <--- new insert line by me on 11.01.2021
  value |= Wire.read();
  value = value << 8;     // <--- new insert line by me on 11.01.2021
  value |= Wire.read();
  // MANUFACTURER
  if (field == 0) return (value >> 12) & 0xFF;
  // PRODUCT ID
  if (field == 1) return value & 0x0FFF;
  // DENSITY
  if (field == 2) return (value >> 8) & 0x0F;
  return 0;
}

RobTillaart, 2021-01-11:
Thanks for noticing,

Added code highlighting around your post.
I will investigate.

Does the remaining part of the library work?

RobTillaart, 2021-01-11:
You are 100% right, the shift is missing.
I will prepare a fix asap.

RobTillaart, 2021-01-11:
Fixed in 0.2.3

closed 2021-01-11 by RobTillaart