RobTillaart/FRAM_I2C #7: Stewrat

Stewrat007, 2021-02-05:
I have been using FRAM_I2C, on a ESP32-WROVER with a I2C MR44V100A (128Kx8) FRAM, for some time now without any problems whatsoever. However with the release of FRAM_I2C version 0.3.0,  PlatformIO has refused to update from version 0.2.2 returning an "Update to 0.3.0 (Incompatible)" message. 

When I look a little deeper into the problem, it appears that an error ('PCF8574' has not been declared) is occurring at line 37 of FRAM.cpp. At present I do not use a PCF8574.

Thank you for such a great library, 

Stewart

RobTillaart, 2021-02-05:
@Stewrat007 
Oops that sounds like something very incorrect. I'll have a look.

RobTillaart, 2021-02-05:
Copy and paste error (mea culpa) and I did not test the code on ESP32 so it did not pop up.
Also the Arduino CI build does not support ESP32 yet afaik. 
Fix is underway.

RobTillaart, 2021-02-05:
Fix tested, fixed again, tested again, now compiles with ESP32Wrover (no hw nearby to test it)
When build is done I will merge it and make a new release 0.3.1

RobTillaart, 2021-02-05:
0.3.1 version released, should fix the issue.

Thanks for pointing out, and if there are other issues please let me know!
Rob

commit a16a9c09 (2021-02-05): fix #7 typo with ESP begin (#8)

* fix #7 typo with ESP begin
closed 2021-02-05 by RobTillaart