RobTillaart/FRAM_I2C #11: getSize returns wrong value

ryangriggs, 2022-02-08:
I tested getSize() with the MB85RC64TAPNF chip which is 64k in size and getSize returns "8".
I can confirm that the chip is actually 64k by writing and reading from address 0xFFFF.

RobTillaart, 2022-02-08:
Thanks for reporting this issue. 
Will look into it asap,
Do the other functions work as expected?

RobTillaart, 2022-02-08:
Found the datasheet - https://nl.mouser.com/datasheet/2/1113/MB85RC64TA_DS501_00044_3v2_E-2329194.pdf

RobTillaart, 2022-02-08:
According to that datasheet **getSize()** should return 3 and not 8
So it might be a slightly other (older?) type without the density bits..

What does the chip return on **getManufacturerID()** and **getProductID()** calls?
(top of the output of testFRAM.ino sketch)

RobTillaart, 2022-02-08:
OK, it took some time to wake up.😁

**getSize()** does return the number of kiloBYTE, and the number 64 in the type of the FRAM is the number of kiloBIT.
64 kilobit == 8 kilobyte.

The reason you can write to address 0xFFFF is that the device just truncates the address bits.
The library does not protect address range 

To see this you need to write 0x00 to all addresses.
Then write 0xFF to 0xFFFF
Then read all address and see where the 0xFF pops up.

If my hypothesis is correct 0xFF should pop up at 0x1FFF - 0x3FFF - 0x5FFF - 0x7FFF - 0x9FFF - 0xBFFF - 0xDFFF and 0xFFFF

RobTillaart, 2022-02-08:
I will already improve the documentation a bit to make the **getSize()** return value more explicit.

ryangriggs, 2022-02-08:
Thank you very much for a fast reply!  As you pointed out, I incorrectly assumed that the getSize function should return 64, but indeed 8 makes sense had I read the docs more carefully to realize it returns BYTES not BITS.  I apologize for any confusion and appreciate you taking the time to clarify this.

The chip returns manufacturer ID 0x0A, Product ID 0x0358

Thank you again for a fast and helpful response!

RobTillaart, 2022-02-08:
you're welcome. took some time for me too to see the light.

I will update the library (mostly documentation) today, so it is more explicit.

> The chip returns manufacturer ID 0x0A, Product ID 0x0358

looks perfect!

If there are no further issues you may close this one

closed 2022-02-08 by ryangriggs

RobTillaart, 2022-02-08:
FYI
Pull Request is opened (develop branch) with updated documentation and function **getSizeBytes()**.