RobTillaart/FRAM_I2C #13: add readObject, writeObject APIs

WestfW, 2022-03-15:
You should ~~steal~~ implement the readObject and writeObject methods from Adafruits FRAM_I2C library, only implementing them properly with the multibyte read and write primitives you already have.
See https://forum.arduino.cc/t/re-fast-data-logging-using-fram/968962/25

RobTillaart, 2022-03-16:
Thanks for the issue, will look into this later this week.

RobTillaart, 2022-03-16:
@WestfW 
Created a new branch - https://github.com/RobTillaart/FRAM_I2C/tree/develop

added two template functions:

- **uint16_t writeObject(uint16_t memaddr, T &obj)** writes an object to memaddr (and following bytes). Returns memaddr + sizeof(obj) to get the next address to write to.
- **uint16_t readObject(uint16_t memaddr, T &obj)** reads an object from memaddr and next bytes. Returns memaddr + sizeof(obj) to get the next address to read from.

Please verify

WestfW, 2022-03-18:
(I don't actually have an FRAM to confirm with.  Issues was submitted based on the forum conversation.)

RobTillaart, 2022-03-18:
Thanks, as my tests showed no problems I will merge later today.

RobTillaart, 2022-03-18:
Merged - 0.3.4

closed 2022-03-18 by RobTillaart