RobTillaart/FRAM_I2C #17: sleep mode

mbmorrissey, 2022-04-02:
Hi,

Are you considering adding suport for sleep?

(I've tried based on
https://www.fujitsu.com/jp/group/fsm/en/documents/products/fram/lineup/MB85RC1MT-DS501-00027-4v0-E.pdf
page 12, but am misunderstanding something!)

[I'm an infrequent github user, sorry if an issue like this isn't the right way to ask this question!]

Michael

RobTillaart, 2022-04-02:
Thanks for the issue, so far I had no plans yet. There were not much requests yet.
It might take a few weeks before I can investigate as I got pretty ill yesterday

mbmorrissey, 2022-04-02:
Thanks for replying about this.  I hope you recover quickly.  Michael

RobTillaart, 2022-04-03:
Looked at the datasheet and one need to send
 0xF8 ADDR 0x8B  (something like that)

So the FRAM seems to listen on a 2nd address which is always F8 and if the following byte matches its 1st address it knows which FRAM to go to sleep.

It might be that this 2nd address pops up on an i2c scanner but as it is above 127 I expect no scanner will test for it.
(enough for today)

mbmorrissey, 2022-04-03:
Hi Rob,

Thanks for looking at it.  I don't want to bug you while you are unwell, so please don't feel pressed to worry about this.

I had tried to follow these instructions from the datasheet, but I've clearly misunderstood something.  

I've had some success using Wire to 'manually' read and write to various devices, but I'm a bit stumped by this one, even though think I've basically followed the flow of things in your source and basically see how it relates to information in the datasheet.  But one thing I don't understand is why an address of 0x50 works, when according to the manual, it looks like the MB85RC1MT ic would be expecting 0x50 << 1 (1010000)?

My attempts have been this, but it shows no sign of a drop in current consuption (measured with Nordic PPK II):

void fram_sleep(){
  Wire.beginTransmission(0xF8);
  Wire.write(0x50); // also have tried  Wire.write(0x50 << 1);
  Wire.write(0x86);  // also have tried Wire.write(0x8B);
  Wire.endTransmission(); 
}
void fram_wake(){
  Wire.beginTransmission(0x50); // ??  Wire.beginTransmission(0x50 << 1);
  delay(10);
  Wire.endTransmission(); 
}

I'm sure I'm being quite naive in some way - my understanding of how to use Wire is quite rudimentary.

Also - please

mbmorrissey, 2022-04-03:
oops - accidentally deleted a bit at the end - was just goign to say please don't worry about this until you are able, and thanks for the pointers so far. Michael

RobTillaart, 2022-04-03:
No you do not press me at all, being ill is booooring.

If you want to learn I2C go to the site of Nick Gammon. His site will boost your knowledge.
 google his name + i2c

mbmorrissey, 2022-04-05:
Thanks for the tip - I’ve studied his information on achieving low power consumption.  I didn’t realise he had other materials like this too!  Hopefully I can carve out some time this week to go through this page in detail.

Michael

RobTillaart, 2022-04-12:
@mbmorrissey 
created branch - https://github.com/RobTillaart/FRAM_I2C/tree/develop - which implements a **sleep()** command.

According to datasheet every normal call will **wake up** the device.

Can you test it works?  
e.g by means of a power test?  some numbers would be great to add to the documentation.

RobTillaart, 2022-04-14:
Found some time to do a test
- stand by: 10.21 uA
- fram.clear():  93.49 uA

Sleep command is not working :(   need to investigate further.

mbmorrissey, 2022-04-14:
OK, thanks for letting me know.

Glad it’s not just me (I gather you got my power consumption test yesterday—it doesn’t seem to be associated with this email chain, but it is in my outbox), but it would have been more convenient if it was just some mistake on my part!

Michael

mbmorrissey, 2022-04-14:
> @mbmorrissey<https://github.com/mbmorrissey>
> Found the cause why my FRAM module is not sleeping. it just does not support it. So I checked all data sheets I have and came to the following list below. Will be added to the readme.md file asap.
>
> As I have no "sleep supporting" FRAM's I can only publish the sleep() and wakeup() as experimental and hope that in time someone will have the right device and will verify / confirm the working (or even fix it).
>
> TYPE    SIZE    SLEEP (datasheet)       CURRENT CONFIRMED
> MB85RC04        512     N       -       N
> MB85RC16        2 KB    N       -       N
> MB85RC64T       8 KB    Y Page 11       4 uA    N
> MB85RC128A      16 KB   N       -       N
> MB85RC256V      32 KB   N       -       N
> MB85RC512T      64 KB   Y Page 12       4 uA    N
> MB85RC1MT       128 KB  Y Page 12       4 uA    N

I do have the MB85RC1MT.  I’ll re-do my test, with another chip if I can find a prototyping board for it, and report back to you (probably tomorrow).  Michael

RobTillaart, 2022-04-21:
@mbmorrissey 
> I do have the MB85RC1MT. I’ll re-do my test, with another chip if I can find a prototyping board for it, and report back to you (probably tomorrow). 
Michael On 14 Apr 2022, at 13:01, 

Any progress made?

mbmorrissey, 2022-04-21:
I’m sorry - I got sucked into something else at work and have not been able to play with electronics.  I’ll try to get to the test tomorrow.  Sorry I didn’t mention I’d not get to it as I said.  Michael

RobTillaart, 2022-04-21:
No problem, just asking.

mbmorrissey, 2022-05-02:
Hi Rob,

I’m sorry it took me so long to get back to testing.  I have now tested with a MB85RC1MT FRAM chip.  Reading with my Nordic PPK II, it sleeps 3.5 & 3.6 uA (tested two separate chips) and draws 12.2 & 11.7 uA in standby.  Would it be useful to have anything further about the conditions of my test?

Again, thanks for incorporating this, and sorry for my delay in getting to testing.

Michael

RobTillaart, 2022-05-02:
@mbmorrissey 
Thanks for this measurement, You showed the sleep function works as intended. 
At the moment I have no further questions,

RobTillaart, 2022-05-02:
@mbmorrissey

After reading the readme again one question popped up.
You have a MB85RC1MT FRAM ==> 128 KB FRAM
As the library does use 16 bit addressing does this library work for that FRAM?
I think you cannot write to the upper half of the FRAM? Or can you?

If not can you please create a new issue for that?
Solution would be to increase the internal address administration to **uint32_t**

RobTillaart, 2022-05-02:
@mbmorrissey 
Can you measure standby  and **clear()** current?  (low prio)

mbmorrissey, 2022-05-02:
Hi,

No problem.  Here is the result of running your FRAM_clear example:

13:23:13.657 -> FRAM_LIB_VERSION: 0.3.6
13:23:13.657 -> BYTES  : 1024
13:23:13.752 ->
13:23:13.752 -> SPEED  : 100000
13:23:14.294 -> BYTES 1: 520226 ==> 15.88 us/byte
13:23:14.722 -> BYTES 2: 320010 ==> 9.77 us/byte
13:23:15.019 -> BYTES 4: 217630 ==> 6.64 us/byte
13:23:15.311 -> BYTES 8: 166356 ==> 5.08 us/byte
13:23:15.563 -> BYTES 16: 140881 ==> 4.30 us/byte
13:23:15.809 -> CLEAR(): 140876 ==> 4.30 us/byte
13:23:16.030 -> CLEAR(0xFF): 140815 ==> 4.30 us/byte
13:23:16.142 ->
13:23:16.142 -> SPEED  : 200000
13:23:16.395 -> BYTES 1: 262759 ==> 8.02 us/byte
13:23:16.678 -> BYTES 2: 159679 ==> 4.87 us/byte
13:23:16.896 -> BYTES 4: 107498 ==> 3.28 us/byte
13:23:17.079 -> BYTES 8: 81736 ==> 2.49 us/byte
13:23:17.224 -> BYTES 16: 68635 ==> 2.09 us/byte
13:23:17.401 -> CLEAR(): 68327 ==> 2.09 us/byte
13:23:17.581 -> CLEAR(0xFF): 68318 ==> 2.08 us/byte
13:23:17.691 ->
13:23:17.691 -> SPEED  : 400000
13:23:17.949 -> BYTES 1: 262772 ==> 8.02 us/byte
13:23:18.203 -> BYTES 2: 159658 ==> 4.87 us/byte
13:23:18.424 -> BYTES 4: 107499 ==> 3.28 us/byte
13:23:18.609 -> BYTES 8: 81737 ==> 2.49 us/byte
13:23:18.793 -> BYTES 16: 68594 ==> 2.09 us/byte
13:23:18.966 -> CLEAR(): 68315 ==> 2.08 us/byte
13:23:19.142 -> CLEAR(0xFF): 68321 ==> 2.08 us/byte
13:23:19.250 -> done...

I have attached a screenshot of the PPK log as the FRAM_clear example ran.  I can zoom in on any parts you’d like.  For e.g., within the periods with consumption to max 720 uA, average current is about 40 uA (this reading taken in the write that I think corresponds to BYTES 4 at  SPEED: 200000).  Idle draw between the writes is still as in my sleep test, approx 11.7 uA.

If I am correctly interpreting the PPK readout from your sketch, here are the average and max current consumptions during clear(), in uA:

Speed average max
100000 26 34
200000 53 721
400000 48 721

And for fram.clear(0XFF):

Speed average max
100000 27 37
200000 47 721
400000 46 721

re. Your other question about the 1M chip: I have assumed that I can only use half of it (which is totally fine for me for now).  I was figuring it would be healthy for me to try to modify your library myself before asking you .  But I’ll be very happy to put in the request (I see you already have it as potentially high priority, pending request)!

Michael

[cid:F387E912-4D94-4944-B492-BF1036AB653B]

mbmorrissey, 2022-05-02:
To more specifically answer your questions:

On 2 May 2022, at 13:01, Rob Tillaart ***@***.******@***.***>> wrote:

As the library does use 16 bit addressing does this library work for that FRAM?

Yes - it does basically work.  I can write data and verify it by reading it back.

I think you cannot write to the upper half of the FRAM? Or can you?

I’m pretty sure I’m only accessing the bottom half.  I presume an attempt to write to the top half would simply over-write corresponding locations in the accessible half, but I have not specifically done anything to verify that.

Michael

mbmorrissey, 2022-05-02:
Sorry I didn’t gather my thoughts better and just send a single email with information in a logical order!

Here is a test, and, yes, it seems to me that accessing the top half just over-writes or reads the bottom half.  Here is my sketch:

#include "FRAM.h"
#include <Wire.h>

FRAM fram;

uint32_t sizeInBytes = 0;
byte testval = 0;

void setup() {

  Serial.begin(9600);
  Wire.begin();
  delay(400);

  int rv = fram.begin(0x50);
  if (rv != 0)
  {
    Serial.print("INIT ERROR: ");
    Serial.println(rv);
  }

  Serial.println();
  Serial.println(__FUNCTION__);

  Serial.print("ManufacturerID: ");
  Serial.println(fram.getManufacturerID());
  Serial.print("     ProductID: ");
  Serial.println(fram.getProductID());
  Serial.print("    memory size: ");
  Serial.println(fram.getSize());

  Serial.println("***************");

  byte test;

  delay(50);
  Serial.println("writing 1 to memory location 1");
  fram.write8(1, 1);

  delay(50);
  Serial.println("reading from memory location 1");
  test =  fram.read8(1);
  Serial.println(test);

  delay(50);
  Serial.println("writing 2 to memory location 65537");
  fram.write8(65537, 2);

  delay(50);
  Serial.println("reading from memory location 65537");
  test =  fram.read8(65537);
  Serial.println(test);

  delay(50);
  Serial.println("reading from memory location 1");
  test =  fram.read8(1);
  Serial.println(test);

}

void loop() {
}

Which outputs:

14:11:52.517 -> ManufacturerID: 10
14:11:52.558 ->      ProductID: 1880
14:11:52.558 ->     memory size: 128
14:11:52.583 -> ***************
14:11:52.619 -> writing 1 to memory location 1
14:11:52.657 -> reading from memory location 1
14:11:52.657 -> 1
14:11:52.691 -> writing 2 to memory location 65537
14:11:52.729 -> reading from memory location 65537
14:11:52.767 -> 2
14:11:52.801 -> reading from memory location 1
14:11:52.839 -> 2

Michael

On 2 May 2022, at 14:05, Michael Morrissey ***@***.******@***.***>> wrote:

To more specifically answer your questions:

On 2 May 2022, at 13:01, Rob Tillaart ***@***.******@***.***>> wrote:

As the library does use 16 bit addressing does this library work for that FRAM?

Yes - it does basically work.  I can write data and verify it by reading it back.

I think you cannot write to the upper half of the FRAM? Or can you?

I’m pretty sure I’m only accessing the bottom half.  I presume an attempt to write to the top half would simply over-write corresponding locations in the accessible half, but I have not specifically done anything to verify that.

Michael

RobTillaart, 2022-05-02:
Thanks, I updated the readme.md with your numbers and will merge the sleep function (0.3.6)
Thereafter we can continue in the new issue #19

closed 2022-05-02 by RobTillaart