RobTillaart/FRAM_I2C #21: second I2C port on ESP32, working with BM280E not with FRAM

Hans1972, 2022-09-23:
Hello,

I tested my code with an BM280E sensor on I2C2 port the code compiled and worked, now I want to replace the BM280E sensor for an FRAM. I thought this wil be easy one only change adres and name.

This gives me an error, sorry it was easier maybe to copy the errorcode, I can do that later if thats neccesary.

The part of the code where I obviously mis something, I tried a lot of things but cant figure it out, I am busy with this for a whole day.
I am not so good in coding, I do my best, but sometimes I get stuck too long :-)

So I tested this on a ESP32:

#define I2C_SDA              21
#define I2C_SCL              22

// FRAM Memory
#define I2C_SDA_2            16
#define I2C_SCL_2            17

// I2C1
TwoWire I2CPower = TwoWire(0);

// I2C2 for FRAM sensor
TwoWire I2CFRAM = TwoWire(1);
FRAM fram;

// Start I2C communication
I2CPower.begin(I2C_SDA, I2C_SCL, I2C_Freq);
I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);

// You might need to change the FRAM I2C address, in our case it's 0x50
if (!fram.begin(0x50, &I2CFRAM)) {
Serial.println("Could not find a valid FRAM sensor, check wiring!");
while (1);
}

RobTillaart, 2022-09-23:
Thanks for the issue,  if time permits I will try to investigate this weekend.

Hans1972, 2022-09-23:
Thanks, enjoy your weekend!

RobTillaart, 2022-09-24:
At least one problem is in this line
```cpp
if (!fram.begin(0x50, &I2CFRAM)) {
```
as there exist no begin function with these parameter type. So think that explains it. 

I've written a minimal sketch based on your snippet. Can you give it a try?

```cpp
#include "Arduino.h"
#include "FRAM.h"

#define I2C_SDA              21
#define I2C_SCL              22

// FRAM Memory
#define I2C_SDA_2            16
#define I2C_SCL_2            17

#define I2C_Freq            100000

// I2C1
TwoWire I2CPower = TwoWire(0);

// I2C2 for FRAM sensor
TwoWire I2CFRAM = TwoWire(1);
FRAM fram(&I2CFRAM);              //  <<<<<<<<<<<<<<<<<<<<<<<

void setup()
{
  Serial.begin(115200);
  Serial.print("FRAM_LIB_VERSION: ");
  Serial.println(FRAM_LIB_VERSION);

  // Start I2C communication
  I2CPower.begin(I2C_SDA, I2C_SCL, I2C_Freq);
  I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);

  // You might need to change the FRAM I2C address, in our case it's 0x50
  if (!fram.begin(0x50))         //  <<<<<<<<<<<<<<<<<<<<<<<
  {
    Serial.println("Could not find a valid FRAM sensor, check wiring!");
    while (1);
  }
  
  Serial.print("ManufacturerID: ");
  Serial.println(fram.getManufacturerID());
  Serial.print("     ProductID: ");
  Serial.println(fram.getProductID());
  Serial.print("     memory KB: ");
  Serial.println(fram.getSize());
}

void loop()
{
}

// -- END OF FILE --

```
Should print something like
```
ManufacturerID: 10
     ProductID: 1296
   memory size: 32
```

RobTillaart, 2022-09-24:
FYI, I created a develop branch which fixes another issue - no relation to yours -

Hans1972, 2022-09-24:
Thanks, I tried to upload this sketch but its hanging on: I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);

Gives me the next errors:
  
C:\Users\..\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\Wire\src/Wire.h:79:10: note: candidate: 'bool TwoWire::begin(int, int, uint32_t)'
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
          ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\Wire\src/Wire.h:80:10: note: candidate: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'
     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
          ^~~~~
exit status 1
call of overloaded 'begin(int, int, int)' is ambiguous

RobTillaart, 2022-09-24:
```I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);``` is outside the library context, but still.

What is happening is that the compiler cannot decide which of the two versions of the **begin()** function to use. 
What the compiler is missing is type information of the parameters.  **#define** does not make the type explicit.
So better use **const int** or **const uint32_t**  (32 bits unsigned integer) to give the compiler the info it needs.

Try this.

```cpp
#include "Arduino.h"
#include "FRAM.h"

const int I2C_SDA = 21;
const int I2C_SCL = 22;

// FRAM Memory
const int I2C_SDA_2 = 16;
const int I2C_SCL_2 = 17;

const uint32_t I2C_Freq = 100000;

// I2C1
TwoWire I2CPower = TwoWire(0);

// I2C2 for FRAM sensor
TwoWire I2CFRAM = TwoWire(1);
FRAM fram(&I2CFRAM);              //  <<<<<<<<<<<<<<<<<<<<<<<

void setup()
{
  Serial.begin(115200);
  Serial.print("FRAM_LIB_VERSION: ");
  Serial.println(FRAM_LIB_VERSION);

  // Start I2C communication
  I2CPower.begin(I2C_SDA, I2C_SCL, I2C_Freq);
  I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);

  // You might need to change the FRAM I2C address, in our case it's 0x50
  if (!fram.begin(0x50))         //  <<<<<<<<<<<<<<<<<<<<<<<
  {
    Serial.println("Could not find a valid FRAM sensor, check wiring!");
    while (1);
  }
  
  Serial.print("ManufacturerID: ");
  Serial.println(fram.getManufacturerID());
  Serial.print("     ProductID: ");
  Serial.println(fram.getProductID());
  Serial.print("     memory KB: ");
  Serial.println(fram.getSize());
}

void loop()
{
}

// -- END OF FILE --
```

Note:  you can use markdown syntax to enhance your posts - see https://www.markdownguide.org/cheat-sheet/

(having a lunch now, back later this afternoon)

Hans1972, 2022-09-24:
Now it compiled, and give me the next message in the serial monitor:

FRAM_LIB_VERSION: 0.4.0
Could not find a valid FRAM sensor, check wiring! 

That seems strange to me, the first line gives the version number, so it looks that the FRAM memory is responding.
But it stops at the second line, the adres of the fram module is right, I tried your example files and that worked correctly.

By the way, thanks for your time!

RobTillaart, 2022-09-24:
The first line only shows the version of the library.
It is the second line or so after ```setup()```

Unfortunately I have no FRAM module nearby to replicate your hardware setup, but OK.
- the examples work
  - fram = ok
  - wiring is ok
  - address is correct

Lets check what the function returns

Can you try

```cpp
#include "Arduino.h"
#include "FRAM.h"

const int I2C_SDA = 21;
const int I2C_SCL = 22;

// FRAM Memory
const int I2C_SDA_2 = 16;
const int I2C_SCL_2 = 17;

const uint32_t I2C_Freq = 100000;

// I2C1
TwoWire I2CPower = TwoWire(0);

// I2C2 for FRAM sensor
TwoWire I2CFRAM = TwoWire(1);
FRAM fram(&I2CFRAM);              //  <<<<<<<<<<<<<<<<<<<<<<<

void setup()
{
  Serial.begin(115200);
  Serial.print("FRAM_LIB_VERSION: ");
  Serial.println(FRAM_LIB_VERSION);

  // Start I2C communication
  I2CPower.begin(I2C_SDA, I2C_SCL, I2C_Freq);
  I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);

  // You might need to change the FRAM I2C address, in our case it's 0x50
  int x = fram.begin(0x50);
  Serial.println(x, HEX);
  
  Serial.print("ManufacturerID: ");
  Serial.println(fram.getManufacturerID());
  Serial.print("     ProductID: ");
  Serial.println(fram.getProductID());
  Serial.print("     memory KB: ");
  Serial.println(fram.getSize());
}

void loop()
{
}
```

RobTillaart, 2022-09-24:
As the examples all work with the 1st I2C bus, it does not tell anything yet about the 2nd I2C bus.

- are you sure that SDA and SCL are correct - not swapped?

Lets try to check if those pins work for **"Wire0"**

```cpp
#include "Arduino.h"
#include "FRAM.h"

// FRAM Memory
const int I2C_SDA_2 = 16;
const int I2C_SCL_2 = 17;

const uint32_t I2C_Freq = 100000;

TwoWire I2CFRAM = TwoWire(0);
FRAM fram(&I2CFRAM);  

void setup()
{
  Serial.begin(115200);
  Serial.print("FRAM_LIB_VERSION: ");
  Serial.println(FRAM_LIB_VERSION);

  // Start I2C communication
  I2CFRAM.begin(I2C_SDA_2, I2C_SCL_2, I2C_Freq);

  // You might need to change the FRAM I2C address, in our case it's 0x50
  int x = fram.begin(0x50);
  Serial.println(x, HEX);
  
  Serial.print("ManufacturerID: ");
  Serial.println(fram.getManufacturerID());
  Serial.print("     ProductID: ");
  Serial.println(fram.getProductID());
  Serial.print("     memory KB: ");
  Serial.println(fram.getSize());
}

void loop()
{
}
```

RobTillaart, 2022-09-24:
Backgrounder: https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/

Hans1972, 2022-09-24:
Thanks, the last code works! I've read the (backgrounder) link above and tried to figure it out myself with that information before I asked here. I complemented the code in my final code and it works correctly.

RobTillaart, 2022-09-24:
> the last code works! 

OK, means the pins work.

> I complemented the code in my final code and it works correctly.

So you use only one I2C bus, OK
If your problem is solved, you may close the issue.

Hans1972, 2022-09-24:
I use now two I2C, the first part allready worked, I replaced the code:

  // You might need to change the FRAM I2C address, in our case it's 0x50
  if (!fram.begin(0x50))         //  <<<<<<<<<<<<<<<<<<<<<<<
  {
    Serial.println("Could not find a valid FRAM sensor, check wiring!");
    while (1);
  }

by

  // You might need to change the FRAM I2C address, in our case it's 0x50
  int x = fram.begin(0x50);
  Serial.println(x, HEX);

closed 2022-09-24 by Hans1972