RobTillaart/FRAM_I2C #23: Ringbuffer

crami, 2022-10-03:
Hi Rob

I'm very interested in using FRAM as ring-buffer. Have you already started coding in that direction?

Regards

  Matthias

RobTillaart, 2022-10-03:
Hi Matthias,

Thanks for opening this issue. I have not taken time to write a ringbuffer yet. Mostly these jobs get done when I need certain functionality in a project myself. Do you have specific requirements or data types in mind?

As I am almost through my open issues and PR's etc, I can give it a try this evening.

RobTillaart, 2022-10-03:
Did you have a look at - FRAM_logging.ino ?
it is a basic ring buffer for logging

RobTillaart, 2022-10-04:
first version - https://github.com/RobTillaart/FRAM_I2C/tree/ringbuffer

how to use see the example - https://github.com/RobTillaart/FRAM_I2C/tree/ringbuffer/examples/FRAM_ringbuffer

crami, 2022-10-04:
Sorry had not the chance to check sooner.

I like to store an arbitrary struct into the ring-buffer like this:
```
struct GPSBuffer {
  float lat;
  float lon;
  float speed;
  float alt;
};
```
I like to be able to store into the buffer and read back until it is empty (FIFO)

RobTillaart, 2022-10-04:
No problem, I have enough things to do

As you know the size of the object, the ringbuffer should be a multiple of this size preferably.
You can test this by writing two functions that do the "serialize and deserialize" to the ring buffer.

Something like this, can you try?
```cpp

bool RB_write(GPSBuffer * data)
{
   //  test if there is space 
  if (size - count < sizeof(GPSBuffer)) return false;
  uint8_t *p = (uint8_t) data;
  for (int i = 0; i < sizeof(GPSBuffer); i++)  framRB.write(*p++);
  return true;
}

bool RB_read(GPSBuffer * data)
{
   //  test if there is enough data
  if (count < sizeof(GPSBuffer)) return false;
  uint8_t *p = (uint8_t) data;
  for (int i = 0; i < sizeof(GPSBuffer); i++)  *p++ =  framRB.read();
  return true;
}
```

If this work I can extend the API to support something like this.

crami, 2022-10-04:
Code looks really promising. I started writing some code around it, but hat no time to test it jet.
Am I correct that size and count should be fb.size and fb.count?

It would be cool if the load and save functions would read memory at the beginning or end of the FRAM where all the ringbuffer parameters are stored (size, head, tail, "signature to check it ring buffer exists") so it can be initialized from there on boot.

RobTillaart, 2022-10-04:
updated - https://github.com/RobTillaart/FRAM_I2C/tree/ringbuffer
(build is running...)

RobTillaart, 2022-10-05:
Notes to myself for possible examples.

The current interface allows to write an identifier (typical 1 byte?) in the ringbuffer, followed by an object of that type. 
That would allow the ringbuffer to contain mixed datatypes.  Optional one also writes the size for certain objects e.g. if one want to write a string of text. 

The ringbuffer can also be used to store a configuration this way. It is not a ring buffer, just a persistent buffer.
At the same time it is overkill as the FRAM class can write/read a whole config struct "at once".

RobTillaart, 2022-10-05:
updated - https://github.com/RobTillaart/FRAM_I2C/tree/ringbuffer

Now the buffer can store any object due to using templates. - interface changed slightly, see example

Please give it a try.

RobTillaart, 2022-10-10:
@crami 
Did you find some time to evaluate?

new version is coming including load() and save().

RobTillaart, 2022-10-11:
### Observations

- As **_count** can be derived from **_front** and **_tail** one need only to store two  of them.
- _start is needed to access the meta data so it makes no sense to store this.

So an update is needed.

RobTillaart, 2022-10-12:
> As _count can be derived from _front and _tail one need only to store two of them.

Apparently not true as the tail and front pointer are identical when buffer == full or when buffer == empty.
(alternative is to keep one slot free so front == tail ==> count = 0;  but that means if you want to buffer for 10 objects of size 8 you need to allocate 80  + 8 bytes which is not "user friendly".

RobTillaart, 2022-10-12:
@crami 
New push,  after several tests the ringbuffer behaves stable. 
- A new example is added that has two ringbuffers
- save(), load() and wipe() seem to work as intended

Please give it a try. I will do more tests if time permits

RobTillaart, 2022-10-14:
@crami
Did you find some time to evaluate?

RobTillaart, 2022-10-16:
@crami 
Apparently you do not have time to evaluate. As I did not find anymore bugs I am going to make a PR of the branch and will merge the ring buffer class later today.

RobTillaart, 2022-10-16:
@crami
New release is nearby
- moved code into .cpp file FRAM.h -> FRAM.cpp
- added a separate FRAM_RINGBUFFER.md file
- updated documentation and meta data

RobTillaart, 2022-10-16:
@crami 
Merged ringbuffer branch into master, released 0.4.2

closed 2022-10-16 by RobTillaart