RobTillaart/FRAM_I2C #25: failing to write a float value, but don't know why

ageurtse, 2022-12-27:
Trying to write a floating point value and reading it back. 
But somehow it dosn't work. 

this is my code i use .

i'm i doing sothing wrong. 
the value i'm writing is 31.12

as you can see the writen and readed value's aren't the same as the sended value. 
How can i fix it. 

```
[12:00:53][I][I2C-FRAM:071]: write value 31.120001 to memory addres 
[12:00:58][I][I2C-FRAM:064]: read value 31.000000 from memory addres 1

```
`float readfromfram(uint32_t memaddr)
{
  float value;
  value = fram.read32(memaddr);
  ESP_LOGI(TAG, "read value %f from memory addres %d", value, memaddr);
  return (value);
}

void writetofram(uint32_t memaddr, float value)
{
  fram.write32(memaddr, value);
  ESP_LOGI(TAG, "write value %f to memory addres ", value, memaddr);
  // ESP_LOGI(TAG, "to memory adres %d ", memaddr);
}`

RobTillaart, 2022-12-27:
Thanks for reporting the issue.
Have no computer nearby (phone only) to investigate.
Will get back as soon as I am operational again.

ageurtse, 2022-12-27:
I think i found the solution myself, a float can't be written to an u_int32_t it has to be writtenand reading as an object.
it know looks like it is working.

RobTillaart, 2022-12-27:
Yes floats are not directly supported, **writeObject()** is the way. As float is a basic datatype it might be usefull to have a **writeFloat(address, float)** to simplify the usage. And a read too of course.

ageurtse, 2022-12-27:
That could be nice, but with the use of writeobject and readbject it does the trick also.

RobTillaart, 2022-12-28:
@ageurtse 
Working on a 0.4.3 release with writeFloat() and readFloat() support in the develop branch.
I encounter problems with build-ci which are under investigation so I will not merge soon.
But you are welcome to give the code a test drive (unit tests are OK).

ageurtse, 2022-12-28:
would love to try it out from within esphome, but don't know how to load the develop branche so i could test it. 
after i found the solution for that i try and report back. 

esphome is based on platformio 
currently i add i like this 

tryed instate of 0.4.2 develop
```
 libraries:
    - "Wire"
    - "robtillaart/FRAM_I2C@^0.4.2"
```

RobTillaart, 2022-12-28:
https://github.com/RobTillaart/FRAM_I2C/tree/develop
Green code button ==> local tab ==> download zip

RobTillaart, 2022-12-28:
The develop branch is not located on platformio so think it will be beyond trivial.
(not familiar with esphome  ==> need to plan some R&D  time...

ageurtse, 2022-12-28:
> The develop branch is not located on platformio so think it will be beyond trivial. (not familiar with esphome ==> need to plan some R&D time...

if you are into home automation, you should realy look at [esphome](https://esphome.io/) and [homeassistant](https://www.home-assistant.io/)

ageurtse, 2022-12-28:
managed to install the develop tree, reading and writing a float looks good to me. 
only thing for me, a float is 32 bit, why do you check for the size ?
this is because if i write a value to the second position it is in my case on the 5th byte.

RobTillaart, 2022-12-28:
I use the sizeof() to minimize the "magic numbers"
The call sizeof(fliat) is self documenting so it provides the reader of the code the meaning of the parameter.

ageurtse, 2022-12-28:
Realy would like to try but don’t now how to select the development enviroment from within esphome (platformio)

closed 2022-12-28 by ageurtse

ageurtse, 2022-12-28:
I managed to load the library and it all run’s fine.

so i see no need to not push it to the master

RobTillaart, 2022-12-28:
Thanks for testing, will merge tomorrow, 
Still I need to have my build-CI running 100% as that saves me a lot of time .

RobTillaart, 2022-12-29:
@ageurtse 
- added writeDouble() and readDouble() to the develop branch. This is for the boards that support 8 byte doubles.
- created a PR
- escalated the bug in the build-ci to the "build team"