RobTillaart/FRAM_I2C #33: esphome port or I2C/GPIO abstraction

sharkydog, 2023-04-22:
Hello,
I am planning to use this in esphome, but I want it to work on esp-idf framework as well.
So, obviously I will have to port it to esphome universe (components, i2cdevice, etc).

Are there any plans to do something similar?
Like, take out low level i2c write out of the main class, so an esphome (or other) integration could be implemented through a small bridge class.

I am not a fluent C++ programmer (I do PHP and JS well), but I've managed to frankenstein working code for some other "side projects" and I think I can port this one.
But, I don't want to do double work for something that is already in the pipeline and probably implemented better than I can.
I'd rather help with the pipeline and wait than do the work and scrap it because upstream did it better :)

RobTillaart, 2023-04-22:
HI, thanks for the question,

> Are there any plans to do something similar?

No, not in my planning. There is one other esphome issue/request, to port a library to that "platform". However keeping all my libraries in shape already takes substantial time. 

> Like, take out low level i2c write/read out of the main class, so an esphome (or other) integration could be implemented through a small bridge class.

Sounds like a plan.

> But, I don't want to do double work for something that is already in the pipeline ...

There is no such thing in my pipeline, so definitely no double work.
If you share your code here I can have a look (if time permits)

sharkydog, 2023-04-22:
Ok, then I proceed.

RobTillaart, 2023-04-23:
FYI, I am currently working on an experimental class to store static text in FRAM. 

The idea is to put translation tables in FRAM, so applications can switch e.g. between Portugese and France and English. 
Might be useful in ESPHOME context?

The class can also be used to store a lot of strings to save RAM,
I have a short example to store the names of the chemical elements in FRAM.

sharkydog, 2023-04-24:
Maybe, not having to reflash language maps every time a tiny little thing (like a sensor filter) has to be changed, could be a good thing.
So far, at least for me, FRAM is most useful to preserve states and counters through power cycles and reflashes.
For that, I simple byte-by-byte access is enough.

sharkydog, 2023-04-25:
This can be closed, unless you like to keep it open for some reason.

RobTillaart, 2023-04-25:
>  Or ... you have otherwise not usable time 

Time is my scarcest resource

```cpp
void FRAM::_writeBlock(uint16_t memaddr, uint8_t * obj, uint8_t size)
{ 
  esphome::i2c::WriteBuffer buff[size+2];
    
  buff[0].data = (const uint8_t *)(memaddr >> 8);
  buff[0].len = 1;
  buff[1].data = (const uint8_t *)(memaddr & 0xFF);
  buff[1].len = 1;
  
  uint8_t * p = obj; uint8_t j = 2;
  for (uint8_t i = size; i > 0; i--)
  {
    buff[j++].data = (const uint8_t *)*p++;
  }
  buff[j].len = 1;   //   <<<<<<<<<<<<<   would be overwritten in the loop.  value of 1 ==> size? or size +2?
  
  //esphome::i2c::I2CDevice::bus_  
  //esphome::i2c::I2CDevice::address_
  bus_->writev(address_, buff, size+2, true);
}

closed 2023-04-25 by RobTillaart

sharkydog, 2023-04-25:
`buff[n].len` is the size of `buff[n].data`

It's used like this (parts of
```c++
// esphome::i2c::ArduinoI2CBus
// https://github.com/esphome/esphome/blob/dev/esphome/components/i2c/i2c_bus_arduino.cpp
ErrorCode ArduinoI2CBus::writev(uint8_t address, WriteBuffer *buffers, size_t cnt, bool stop) {
  // ...
  wire_->beginTransmission(address);
  for (size_t i = 0; i < cnt; i++) {
    const auto &buf = buffers[i];
    if (buf.len == 0)
      continue;
    size_t ret = wire_->write(buf.data, buf.len);
    written += ret;
    if (ret != buf.len) {
      ESP_LOGVV(TAG, "TX failed at %u", written);
      return ERROR_UNKNOWN;
    }
  }
  uint8_t status = wire_->endTransmission(stop);
  // ...
}

sharkydog, 2023-04-25:
I wanted to leave intact as much of your code as possible, so diff and merge would be easier later.

sharkydog, 2023-04-27:
One last thing ad I am leaving you alone.
As I have bad understanding of C++ in general and pointers mostly, I've been reading a bit.
So, this
```c++
  esphome::i2c::WriteBuffer buff[size+2];
    
  buff[0].data = (const uint8_t *)(memaddr >> 8);
  buff[0].len = 1;
  buff[1].data = (const uint8_t *)(memaddr & 0xFF);
  buff[1].len = 1;
```
is not going to work as I expect, `buff.data` should be an address of the value and not the value.

Option one, various forums and guides say "endianness aware code is bad"
```c++
// uint16_t memaddr
// memaddr 0xABCD, first byte 0xCD, next byte 0xAB, little-endian
uint8_t * memaddrPtr = reinterpret_cast<uint8_t *> (&memaddr);
buff[0].data = memaddrPtr + 1;
buff[1].data = memaddrPtr;
```

Option two
```c++
uint8_t addr[] = { (memaddr >> 8), (memaddr & 0xFF) };
buff[0].data = &addr[0];
buff[1].data = &addr[1];
```

RobTillaart, 2023-04-27:
Not familiar with the ESPHOME API, but I would go one level lower.

That would result in something like this.

```cpp
void FRAM::_writeBlock(uint16_t memaddr, uint8_t * obj, uint8_t size)
{
  //  write the memory address
  uint8_t data= memaddr >> 8;
  wire_->write(&data, 1);
  data = memaddr & 0xFF;
  wire_->write(&data, 1);

  // write the blok
  uint8_t *p = obj;  //  set at start of block
  for (uint8_t i = 0; i < size; i++, p++)
  {
     wire_->write(*p, 1);  //  write the content of byte where p points to.
  }
}
```

sharkydog, 2023-05-01:
> It’s alive, it’s ALIVE…!
_-- Victor Frankenstein_

So, I settled on this:
```c++
void FRAM::_writeBlock(uint16_t memaddr, uint8_t * obj, uint8_t size)
{
  esphome::i2c::WriteBuffer buff[2];
  uint8_t addr[] = { (uint8_t)(memaddr >> 8), (uint8_t)(memaddr & 0xFF) };
  
  buff[0].data = addr;
  buff[0].len = 2;
  buff[1].data = obj;
  buff[1].len = size;
  
  bus_->writev(address_, buff, 2, true);
}
```
```c++
void FRAM::_readBlock(uint16_t memaddr, uint8_t * obj, uint8_t size)
{
  uint8_t addr[] = { (uint8_t)(memaddr >> 8), (uint8_t)(memaddr & 0xFF) };
  bus_->write(address_, addr, 2, true); // stop=false?
  bus_->read(address_, obj, size);
}
```
And I am testing it like this:
```c++
uint8_t bytes_write[3] = {0xA5,0xB5,0xC5};
fram->write(0x0001,bytes_write,3);

uint8_t bytes_read[2];
fram->read(0x0002, bytes_read, 2);
```

The thing I don't get:
The datasheet for MB85RC256V says "Sequential Read" follows "Random Read", which has no "STOP condition" between memory address write and read "START condition", but ```FRAM::_readBlock``` sends "STOP", in your code as well.
```c++
void FRAM::_readBlock(uint16_t memaddr, uint8_t * obj, uint8_t size)
{
  _wire->beginTransmission(_address);
  _wire->write((uint8_t) (memaddr >> 8));
  _wire->write((uint8_t) (memaddr & 0xFF));
  _wire->endTransmission(); // <===== stop=false?
  _wire->requestFrom(_address, size);
  uint8_t * p = obj;
  for (uint8_t i = size; i > 0; i--)
  {
    *p++ = _wire->read();
  }
}
```
So, read should fail, but doesn't, why?

[parts of esphome's i2c_bus.h](https://github.com/esphome/esphome/blob/2023.4.0/esphome/components/i2c/i2c_bus.h)
```c++
// esphome::i2c
struct ReadBuffer {
  uint8_t *data;
  size_t len;
};
struct WriteBuffer {
  const uint8_t *data;
  size_t len;
};

// class esphome::i2c::I2CBus
  virtual ErrorCode read(uint8_t address, uint8_t *buffer, size_t len) {
    ReadBuffer buf;
    buf.data = buffer;
    buf.len = len;
    return readv(address, &buf, 1);
  }
  virtual ErrorCode readv(uint8_t address, ReadBuffer *buffers, size_t cnt) = 0;
  virtual ErrorCode write(uint8_t address, const uint8_t *buffer, size_t len, bool stop) {
    WriteBuffer buf;
    buf.data = buffer;
    buf.len = len;
    return writev(address, &buf, 1, stop);
  }
  virtual ErrorCode writev(uint8_t address, WriteBuffer *buffers, size_t cnt, bool stop) = 0;
```

RobTillaart, 2023-05-01:
Well done, looks pretty simple in the end.

> So, read should fail, but doesn't, why?

Datasheet says how it should be used, but that does not mean there are "loopholes". 
```cpp
  _wire->beginTransmission(_address);
  _wire->write((uint8_t) (memaddr >> 8));
  _wire->write((uint8_t) (memaddr & 0xFF));
  _wire->endTransmission(); // <===== stop=false?
```
This part sets the memory address to use . Apparently the stop condition does not alter the internal register/pointer it uses for the next command, which is the **readFrom()**. 

Another point could be multi-master mode, in which multiple masters are on the I2C bus. (never worked with).
I assume that after a **stop** other masters may start to send commands as well resulting in interference.
In a multithreading environment this could also happen if two threads try to read from the FRAM. This should always be prevented by adding a queue before the hardware.

Bus as said, never worked with multi master myself, and there are no problems reported on this topic so far.

sharkydog, 2023-05-05:
It matters for device info (get size), just tested.

sharkydog, 2023-05-05:
And the product.
https://github.com/sharkydog/esphome-fram

RobTillaart, 2023-05-05:
Super!