RobTillaart/FRAM_I2C #34: [Question]

N0ury, 2023-04-23:
After recovering from deepsleep fram.begin produces the error:
[    68][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1

This doesn't happen after a hard reset, only when waking from deepsleep.

Any idea?

I use a begin with pins only int rv = fram.begin(I2C_SDA, I2C_SCL);
This because I use non standard pins on an ESP32-C3
Before adding FRAM management all was good.
My code is too big to paste it here.
It must be related to the use of FRAM, but how?

RobTillaart, 2023-04-23:
> Any idea?

no, sorry

Does it also happen when you use standard pins?

> My code is too big to paste it here.

Create a minimal version that shows the problem. Should not be to difficult.

RobTillaart, 2023-04-23:
Some thoughts popped up.

- is the FRAM powered during the deep sleep of the ESP? 
- Some types of FRAM have a sleep mode, and need up to 400 usec to recover (search for TREC in datasheet).
  - it might be slower to wake up than the ESP32.

N0ury, 2023-04-23:
I'm writing a minimal version.
No, FRAM is not powered during deepsleep.
When the ESP wakes up a pin is put as OUTPUT and HIGH to poweron FRAM.
This is the same when there's a hard reset.
I'm going to try addind a delay before posting the minimal version.

N0ury, 2023-04-23:
You were right!
After adding a 1 second delay it works fine.
It works also with 400ms.
I understand that for the first run pins are pulled up before entering in setup.
That's why the problem appears only after the first deepsleep.
I haven't found TREC in datasheet.
Anyway thanks for your help!

RobTillaart, 2023-04-23:
> I haven't found TREC in datasheet.

might be called different in different datasheets. 
You should be able to find this kind of wake up delays in the datasheet.

You may close the issue if solved

closed 2023-04-23 by N0ury