RobTillaart/FRAM_I2C #35: MB85RC04 FRAM9

GWiking, 2023-04-28:
Hi!

First of all, I highly appreciate the time you spend doing good libraries!

I have tried MB85RC04 FRAM (with ESP32) with FRAM9_Test example. 
Changes before run:
It is not possible to get the sizeInBytes = fram.getSizeBytes(); so I have set it to sizeInBytes = 100 and changed the printout, of read,  to Serial.println(**x**); instead of  Serial.println(**value**); to actually see what is read.

The result I get is:

15:03:18.301 -> Write	95	48
15:03:18.301 -> Write	96	49
15:03:18.301 -> Write	97	49
15:03:18.301 -> Write	98	50
15:03:18.301 -> Write	99	50
15:03:18.301 -> Read	0	50
15:03:18.301 -> Read	1	50
15:03:18.301 -> Read	2	50
15:03:18.301 -> Read	3	50
15:03:18.301 -> Read	4	50
15:03:18.301 -> Read	5	50
15:03:18.301 -> Read	6	50
15:03:18.333 -> Read	7	50

I have tried to only write one address and the result is the same.

RobTillaart, 2023-04-28:
Thanks for the issue.
Might take a few days before I can dive into this. (Too busy)
As I have no such FRAM to test it might ask you to do test runs.

GWiking, 2023-04-28:
Sure, no problem at all.

GWiking, 2023-04-28:
Im pretty sure that it is an addressing error and that I read address 0 every time.  When I tried setting i.e 0x1000 and  0x2000 instead of 0x01 0x02, it is possible to write and read those addresses with different values. I hope this helps out.

RobTillaart, 2023-04-28:
> It is not possible to get the sizeInBytes = fram.getSizeBytes(); 

The size of the FRAM9 == 512 bytes, it is hard coded so it should return 512 
Ran the testsketch without an FRAM9 and even then it returns:
```
FRAM_LIB_VERSION: 0.5.2
INIT ERROR: -12
   KB  :	0
BYTES  :	512
...
```

So what do you get (first 4-5 lines)?

RobTillaart, 2023-04-28:
Q: you are using which version of the library?  

FRAM9 support is experimental and not tested by me, wrote it from datasheet after a request for FRAM11
See https://github.com/RobTillaart/FRAM_I2C/issues/28

RobTillaart, 2023-04-28:
Q: can you post your modified sketch, might help.

GWiking, 2023-04-28:
Im using latest version.

GWiking, 2023-04-28:
20:23:24.063 -> FRAM_LIB_VERSION: 0.5.2

20:23:24.063 ->    KB  :       0

20:23:24.063 -> BYTES  :       0

20:23:24.174 -> done...

GWiking, 2023-04-28:
//
//    FILE: FRAM9_test.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo FRAM9 memory test
//     URL: https://github.com/RobTillaart/FRAM_I2C

#include "FRAM.h"

FRAM9 fram;

uint32_t start;
uint32_t stop;

uint32_t sizeInBytes = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("FRAM_LIB_VERSION: ");
  Serial.println(FRAM_LIB_VERSION);

  Wire.begin();

  int rv = fram.begin(0x50);
  if (rv != 0)
  {
    Serial.print("INIT ERROR: ");
    Serial.println(rv);
  }

  //  get size
  Serial.print("   KB  :\t");
  Serial.println(fram.getSize());
  Serial.print("BYTES  :\t");
  Serial.println(fram.getSizeBytes());
  sizeInBytes = fram.getSizeBytes();

  delay(100);

  test();

  Serial.println("done...");
}

void loop()
{
}

void test()
{
  //  addr goes from 0 .. 512
  //  value goes from 0.. 255

  // uint8_t value = 0x00;
  // for (uint32_t addr = 0; addr < sizeInBytes; addr++)
  // {
  //   if (addr % 2 == 0) value++;
    fram.write8(0x1000, 0x0A);
    fram.write8(0x2000, 0x0B);
  
    Serial.println(fram.read8(0x1000));
    Serial.println(fram.read8(0x2000));

  //   Serial.print("Write\t");
  //   Serial.print(addr);
  //   Serial.print("\t");
  //   Serial.println(value);
  // }

  // // read back
  // value = 0x00;
  // for (uint32_t addr = 0; addr < sizeInBytes; addr++)
  // {
  //   if (addr % 2 == 0) value++;
  //   uint8_t x = fram.read8(addr);
  //   if (x != value)
  //   {
  //     Serial.print("Read\t");
  //     Serial.print(addr);
  //     Serial.print("\t");
  //     Serial.println(value);
  //   }
  // }

}

//  -- END OF FILE --

GWiking, 2023-04-28:
gives printout:

20:29:50.827 -> FRAM_LIB_VERSION: 0.5.2
20:29:50.827 ->    KB  :	0
20:29:50.827 -> BYTES  :	0
20:29:50.829 -> 10
20:29:50.829 -> 11
20:29:50.829 -> done...

GWiking, 2023-04-28:
And with  FRAM not connected:

20:32:03.635 -> FRAM_LIB_VERSION: 0.5.2
20:32:03.635 -> INIT ERROR: -12
20:32:03.635 ->    KB  :	0
20:32:03.635 -> BYTES  :	512
20:32:04.738 -> 255
20:32:05.739 -> 255
20:32:05.739 -> done...

RobTillaart, 2023-04-28:
Thanks for this data, think I have enough to start to investigate.

```cpp
    fram.write8(0x1000, 0x0A);
    fram.write8(0x2000, 0x0B);
  
    Serial.println(fram.read8(0x1000));
    Serial.println(fram.read8(0x2000));
```

These addresses are way bigger than the 512 bytes. OK interesting,

RobTillaart, 2023-04-28:
Think I found the cause.

- FRAM9 inherits **begin()** from FRAM.
- **begin()** calls **getSize()** 

this is the getSize from FRAM, not from FRAM9, you can verify this by setting a print statement in the getSIze of FRAM.

I will create a develop branch to fix this.

GWiking, 2023-04-29:
Result:
07:24:07.763 -> FRAM_LIB_VERSION: 0.5.3
07:24:07.763 ->    KB  :	0
07:24:07.763 -> BYTES  :	512

Printout for the write read, just to show result:
Changed to printout actually data Serial.println(x);

07:24:08.474 -> Write	508	255
07:24:08.507 -> Write	509	255
07:24:08.507 -> Write	510	0
07:24:08.507 -> Write	511	0
07:24:08.507 -> Read	0	255
07:24:08.507 -> Read	1	255
07:24:08.507 -> Read	2	255
07:24:08.507 -> Read	3	255
07:24:08.507 -> Read	4	255
07:24:08.507 -> Read	5	255

RobTillaart, 2023-04-29:
Looks good! Sounds like it is working now as expected.
Or are there still problems?

I will reference this issue in the readme as "FRAM9 test"

GWiking, 2023-04-29:
Yes still problem with the addresses of the memory please see the printout. It only reads or maybe also writes the same address. Thats why I played around and found that 0x1000 worked but not 0x0001.

RobTillaart, 2023-04-29:
Ok will investigate further after lunch.

RobTillaart, 2023-04-29:
mmm, looks like the same problem as the getSize.
I have added write8() and read8() to the FRAM interface to be sure the right writeBlock() is called.

Can you give it  a try?
If that works I will move the whole interface.

GWiking, 2023-04-29:
Brilliant!

I think You have a debug print in write also "hoela"? 

This is the result write->read

Removed the if and printed out all read values even when matching the write.

12:59:40.030 -> Write	506	254
12:59:40.030 -> hoela
12:59:40.030 -> Write	507	254
12:59:40.030 -> hoela
12:59:40.030 -> Write	508	255
12:59:40.030 -> hoela
12:59:40.030 -> Write	509	255
12:59:40.030 -> hoela
12:59:40.030 -> Write	510	0
12:59:40.030 -> hoela
12:59:40.030 -> Write	511	0
12:59:40.030 -> Read	0	1
12:59:40.030 -> Read	1	1
12:59:40.030 -> Read	2	2
12:59:40.030 -> Read	3	2
12:59:40.030 -> Read	4	3
12:59:40.030 -> Read	5	3
12:59:40.030 -> Read	6	4
12:59:40.030 -> Read	7	4

Thank You very much for Your work!

RobTillaart, 2023-04-29:
Good to hear it is working, it was my mistake regarding inheritance.
I need investigate how to improve or so (template class)

Pushed a cleaned version, without the "hoela"  😎 which I used to see that the correct **writeBlock()** was called.

I won't merge the develop now as I want to look at it later with a fresh pair of eyes. 
And I need to fix the FRAM11 class too, has the same problems. 

Thanks for testing so far!
Can you test the following functions for me?
- getManufacturerID();  //  just interested in the return value
- getProductID();   //  just interested in the return value
- clear()      // as it uses internal size admin.

RobTillaart, 2023-04-29:
Latest push include the fixes for FRAM11, so no changes for FRAM9.

GWiking, 2023-04-29:
I will do the testing on monday, I have no HW available until then.

RobTillaart, 2023-05-01:
@GWiking 
I rewrote the FRAM9/11 implementation. 
Made the low level writeBlock() and readBlock() virtual so the right code is selected runtime. 
Also getSize() became virtual.

Please let me know if the latest version still works.

GWiking, 2023-05-01:
Yes still works :)

Clear OK!

12:58:00.552 -> ManufID:10
12:58:00.552 -> ProdID:16

RobTillaart, 2023-05-01:
Super, thanks for testing. 
Then I will merge later today.
If you have time could you please read the readme.md to see if things sre missing?
I read it too often to see ambiguities or worse missing information.

GWiking, 2023-05-01:
Looks great! Thanks!

closed 2023-05-01 by RobTillaart

RobTillaart, 2023-05-01:
Merged develop and released 0.5.3, 
Thanks for all your testing!

commit b536e637 (2023-05-01): fix #35 - redo FRAM9 and FRAM11- getSize() (#36)

- major redo FRAM9 and FRAM11, see #35
- update FRAM32 to match FRAM
- update readme.md.
- minor edits