RobTillaart/FRAM_I2C #37: Support for Additional FRAM Devices and Addressing Modes

hemangjoshi37a, 2023-05-18:
I would like to request the addition of support for additional FRAM devices and addressing modes in the FRAM_I2C library. This would enhance the library's versatility and compatibility with a wider range of FRAM modules.

Currently, the library supports FRAM devices with 16-bit, 32-bit, 11-bit, and 9-bit addressing modes. However, there are other FRAM devices available in the market with different addressing modes and sizes that are not yet supported.

I propose the following features to be added to the library:

1. Support for FRAM devices with custom addressing modes: Extend the library's capabilities to accommodate FRAM devices with unique addressing modes that deviate from the existing supported modes (16-bit, 32-bit, 11-bit, and 9-bit). This would enable users to interface with a wider variety of FRAM devices and unlock their full potential.

2. Compatibility with additional FRAM sizes: Enhance the library to work seamlessly with FRAM devices of different sizes, including but not limited to the MB85RC128A, which currently requires manual configuration using the `setSizeBytes()` function. Automating the size detection process would simplify usage and improve the overall user experience.

By incorporating these enhancements, the FRAM_I2C library would become more adaptable and accessible to a larger user base. It would empower developers to utilize a wider range of FRAM devices in their projects, ensuring optimal performance and flexibility.

I believe these additions would greatly benefit the library's functionality and increase its usefulness in various applications. Thank you for considering my feature request.

RobTillaart, 2023-05-18:
Thanks for this issue.

The suggestions are interesting.
I have not all different FRAM hardware so it is hard to develop / test. 

Do you have a list of the other FRAM devices?
Preferably a link to the data sheets.
Links to experimental code helps too.

Of course feel free to make a PR .

RobTillaart, 2023-05-26:
As no info is given, issue is closed. Feel free to reopen if needed.

closed 2023-05-26 by RobTillaart