RobTillaart/FRAM_I2C #38: getMetadata() not working on MB85RC64V

pvogt09, 2023-09-02:
I have a MB85RC64V FRAM chip and although I am able to read and write it, the metadata information is not working.
The exact component I am using is [this](https://jlcpcb.com/partdetail/Fujitsu-MB85RC256VPF_GJNERE2/C8740) but the datasheet does not say anything about a metadata address code to send, so I would assume that it does not supply this information.
In https://github.com/RobTillaart/FRAM_I2C/blob/b536e637f21976bb7163ff4fd437c276cda71109/FRAM.cpp#L352 the number of read bytes is always `0` so the function returns `-1`. The datasheet says, that the device only uses 13 bits for the addresses and I am wondering if it would be necessary to have a `FRAM13` class (with fixed memory size like in `FRAM9`) as well for this case although it does not work fundamentally different from the base `FRAM` class?

In addition I am wondering if it would make sense to return the number of written bytes returned by `_wire->write` in all the `write*` methods to give the user the ability to check if something went wrong during the write operation?

sharkydog, 2023-09-03:
1. It won't return -1 because of the return type.
https://github.com/RobTillaart/FRAM_I2C/blob/b536e637f21976bb7163ff4fd437c276cda71109/FRAM.cpp#L344

2. It should not need a new class as it behaves just like 32KiB chip (FRAM class, 2 bytes memory address), while FRAM9, FRAM11, FRAM32 need special treatment because they use device address bits for memory address.

3. I don't think that's necessary.

RobTillaart, 2023-09-03:
@sharkydog 
1. You are right, -1 will be mapped upon 0xFFFF which could cause errors in the calling code. 
Especially getSize(), however it would set the size to 1UL << 0xFFFF which would become 0. 
So not handled as it should, but no harm done. 

getSize() should have an extra test at least be rewritten as:
```cpp
// NOTE: returns the size in kiloBYTE
uint16_t FRAM::getSize()
{
  uint16_t size = 0;
  uint16_t density = _getMetaData(2);
  if (density != 0xFFFF)
  {
    if (0 < density) && (density < 16))  size = (1UL << density);   // only shift for valid values.
  }
  _sizeBytes = size * 1024UL;
  return size;
}
```
Will add a note to include this in some next release.

2. Agree
Had a quick look at the datasheet and it indeed uses 2 bytes for the memory address (13 for address and 3 MS bits 0).
The 3 zero bits are not used otherwise so no new class needed.

BTW The MB85RC64 type is mentioned in the readme.md. T version instead of V version, the T does implement the metadata where the V does not.

> In addition I am wondering if it would make sense to return the number of written bytes returned by _wire->write in all the write* methods to give the user the ability to check if something went wrong during the write operation?

It could make sense, but writing the bytes do not guarantee that the EEPROM is written, only a verify / read back is 100%

sharkydog, 2023-09-04:
You can skip `if (density != 0xFFFF) {}` as you have this `(density < 16)` and 65535 is certainly not lower than 16 :)

But It should also be noted that `FRAM::getManufacturerID()` and `FRAM::getProductID()` may return 0xFFFF

RobTillaart, 2023-09-04:
I will push a develop branch today with these and some more.

RobTillaart, 2023-09-04:
Pull Request created - https://github.com/RobTillaart/FRAM_I2C/pull/41

RobTillaart, 2023-09-07:
Fixed in 0.5.4

closed 2023-09-07 by RobTillaart