RobTillaart/FRAM_I2C #42: FRAM32 and read / write object

geologic, 2023-09-15:
Hi

I'm using FRAM32 library on a MB85RC1MT to read/write STRUCT objects. When compiling i get this error:

undefined reference to `unsigned long FRAM32::readObject<LogDataStructure>(unsigned long, LogDataStructure&)'

The same for writeObject.

Can you please advise?

RobTillaart, 2023-09-15:
Thanks for the issue, 
Will look into it tomorrow as today is "full".

RobTillaart, 2023-09-15:
Could you check if you could read/writeObject of a simple datatype e.g int?

RobTillaart, 2023-09-15:
Hypothesis
In the FRAM class the functions are implemented in .h
In the FRAM32 class the functions are in .cpp.

Moving the implementation to .h file miight solve it
Todo: write text sketch tomorrow

RobTillaart, 2023-09-16:
Problem confirmed => BUG

RobTillaart, 2023-09-16:
@geologic 
- Moving the template functions to the .h file solved the problem for AVR and ESP32.
- Created a develop branch and a PR for version 0.6.0
- Added two examples that use it for FRAM32 and FRAM11 (other derived class)

If you have time, could you verify the develop branch if it works for you?

closed 2023-09-17 by RobTillaart

geologic, 2023-09-19:
Just to let you know that now it's working, thanks!

RobTillaart, 2023-09-19:
Very important feedback!

commit 221e8dc7 (2023-09-17): fix #42 ReadObject, WriteObject (#43)

- fix #42 ReadObject, WriteObject for FRAM32
- add examples
- minor edits