RobTillaart/FRAM_I2C #45: Tested device: FM24C256-G (Ramtron) Works

ifreislich, 2023-10-03:
Rob,

For your supported table: the Ramtron FM24C256-G works with this library. I'm using it in several projects and It's *much cheaper than the Fujitsu devices. Despite the datasheet stating Vcc=4.7-5.5V I've only ever used the device with an ESP32 and ESP8266 at 3.3V.

It does not have a device register so there's no device metadata.
It does not support sleep.

Standby current 100uA
Active current SCL=100kHz 200uA, SCL=400kHz 500uA, SCL=1MHz 1.2mA

RobTillaart, 2023-10-03:
Super!
Thanks for this info / confirmation.
Will update documentation asap

closed 2023-10-04 by RobTillaart

ifreislich, 2023-10-07:
It's 256kbit in 32k * 8.

RobTillaart, 2023-10-07:
Sorry my mistake, will fix asap.

ifreislich, 2023-10-07:
No worries. Thanks for a great library.