RobTillaart/FRAM_I2C #48: Connection to FRAM stopped working with version 0.7.0

tkrippesyuri, 2023-11-15:
Hi,

I'm currently using the FRAM_I2C library on a custom board based on the Adafruit Feather M4 CAN board. Since a couple of days, I'm unable to connect to the FRAM whereas it worked before. Whenever I try to connect to it using "fram.begin(0x51)" it returns me "-12" as an error code. Subsequent tries to read from or write to the FRAM fail as well.

I was able to find out that the connection was possible with version 0.6.1, but not with the latest version 0.7.0 (I updated the library a couple of days ago). Here are some additional information:

Base Board: [Adafruit Feather M4 CAN](https://www.adafruit.com/product/4759)
FRAM: [Fujitsu MB85RC256VPNF-G-JNERE1](https://www.mouser.lu/ProductDetail/Fujitsu-Semiconductor/MB85RC256VPNF-G-JNERE1?qs=HBWAp0VN4RgK2GET9pGgxw%3D%3D))
I2C address: 0x51

I know that it's not much information, but I hope it helps you find the issue. If you need any more information, feel free to reach out.

RobTillaart, 2023-11-15:
Thanks for the issue, I will dive into it later today .

RobTillaart, 2023-11-15:
Is there a specific example that shows the problem. That would allow me to recreate it .

RobTillaart, 2023-11-15:
You need to call Wire.begin() before FRAM.begin().

That is a breaking change, which allows easier support for ESP32 / RP2040 et al, as these have slightly different Wire implementations as the Arduino UNO / Mega

tkrippesyuri, 2023-11-15:
I'm sorry, you're right. In the small example code I have, I forgot to include "Wire.begin()" and in my actual program I called it, but only after calling "FRAM.begin()". Calliing it before actually solved the issue, thank you very much!

closed 2023-11-15 by RobTillaart