RobTillaart/FRAM_I2C #49: FM24V10

ddprog, 2023-12-25:
Hello!

I'm trying to start FM24V10, but it is not detected correctly. Can this be fixed?

[image]

[image]

RobTillaart, 2023-12-25:
Merry Christmas,
I have seen your question and it might take a few days before I can investigate.

ddprog, 2023-12-25:
FM24V10 = 1 Mbit.
I expected to see 128 kiloBytes.

RobTillaart, 2023-12-25:
OK, 
The FM24V10 was not in my test-set when developing the library. 
The library takes **pow(2, density)** as size in KB, which is not true for the FM24V10.
So I need to study the datasheet of more FRAM's from this supplier to see if I can come up with a generic algorithm, lookup table or maybe the user has to set the model.

Given it is 128 KB, do you use FRAM32 class?
Does the library has other problems with this FM24V10?

RobTillaart, 2023-12-25:
Does the FM24V10 datasheet describe the meaning of the density bits ?

RobTillaart, 2023-12-25:
### Notes to myself:

Implement a **getSize()** that uses a lookup table per Manufacturer ID.

Cypress/infineon  = 0x004  - dens 4 = 128 KB   dens 3 = 64 KB  (FM24V05)  =>  (1 << density) * 8KB  ?
Fujitsu = 0x00A => current code

https://nl.farnell.com/c/semiconductors-ics/memory/ferroelectric-ram-fram?brand=infineon&interfaces=i2c

RobTillaart, 2023-12-25:
Created a develop branch with getSize() patched

Or you can replace the **getSIze()** function in **FRAM.cpp** by the function below
(comment the old version)
```cpp
#define FRAM_MANU_FUJITSU     0x0A
#define FRAM_MANU_CYPRESS     0x04

//  DENSITY  Fujitsu data sheet
//  3 =>     MB85RC64 = 64 Kbit.
//  5 =>     MB85RC256
//  6 =>     MB85RC512
//  7 =>     MB85RC1M
//
//  DENSITY  Cypress / Infineon data sheet
//  3 =>     FM24V05 = 64 KByte.
//  4 =>     FM24V10 = 128 KByte.
//
//  NOTE: returns the size in kiloBYTE (0 is read error)
uint16_t FRAM::getSize()
{
  uint32_t value = _getMetaData();
  if (value == 0xFFFFFFFF) return 0;
  
  uint16_t manufacturer = (value >> 12) & 0x0FFF;

  if (manufacturer == FRAM_MANU_CYPRESS)
  {
    uint16_t density = (value >> 8) & 0x0F;
    uint16_t size = (1UL << density) * 8;  //  KB
    _sizeBytes = size * 8192UL;
    return size
  }
  //  default FRAM_MANU_FUJITSU
  uint16_t density = (value >> 8) & 0x0F;
  uint16_t size = (1UL << density) * 1;  //  KB
  _sizeBytes = size * 1024UL;
  return size;
}
```

ddprog, 2023-12-26:
Hello! Thank you for your quick response, there is no way to check your code yet. Yes, I am using the FRAM 32 class. In my project I am planning to use a ring buffer. As soon as I have the opportunity to test it, I will write to you immediately.

ddprog, 2024-01-09:
Hello!
I'm trying your development branch, it detects the microchip correctly. I'll try to improve my program for the write and read test.
[image]
Code:
`    if (fram.begin(FRAM_ADDRESS) == 0)
    {
        ddLog("[INIT] FRAM Start Succeded!", serialMutex);
        isFram = true;
        ddLog(("[INIT] FRAM ManufID: " + (String)(fram.getManufacturerID())), serialMutex);
        ddLog(("[INIT] FRAM ProdID: " + (String)(fram.getProductID())), serialMutex);
        ddLog(("[INIT] FRAM Size: " + (String)(fram.getSize())), serialMutex);
    }
    else
    {
        ddLog("[INIT] FRAM Start Failed!", serialMutex);
        isFram = false;
    }`

RobTillaart, 2024-01-09:
Thanks for verifying the develop branch.
Will do s last review and merge it later today.

Your code, minor remark.
Succeeded is written with three times e.

closed 2024-01-09 by RobTillaart