RobTillaart/FRAM_I2C #51: Bug with reading

ddprog, 2024-01-15:
Hello!

I found a bug with reading from FRAM from the second half of the addresses (1 Mbit). Please look and correct.
[image]

ddprog, 2024-01-15:
Corrected for _addr, it seems to read correctly.

RobTillaart, 2024-01-15:
Thank you for reporting the bug. (not tested well enough apparently)

Code should indeed be:

```cpp
void FRAM32::_readBlock(uint32_t memAddr, uint8_t * obj, uint8_t size)
{
  uint8_t _addr = _address;
  if (memAddr & 0x00010000) _addr += 0x01;

  _wire->beginTransmission(_addr);
  _wire->write((uint8_t) (memAddr >> 8));
  _wire->write((uint8_t) (memAddr & 0xFF));
  _wire->endTransmission();
  _wire->requestFrom(_addr, size);
  uint8_t * p = obj;
  for (uint8_t i = size; i > 0; i--)
  {
    *p++ = _wire->read();
  }
}
```

Will create a PR asap.

RobTillaart, 2024-01-15:
Develop branch + PR created

RobTillaart, 2024-01-15:
#### Note
if memAddr would be e.g. 0x00020001,  would interfere with lower memory addresses too.
Robust guarding solves this too.

Note in readme.md future section to investigate error handling/ reporting wrt invalid memory addresses

commit 1e7a5e01 (2024-01-16): Fix #51 - bug in FRAM32::_readBlock() (#52)

- Fix #51 - bug in FRAM32::_readBlock()
- improve fram32 example to test both low and high addresses
- add memory address guarding for the FRAM32 _readBlock() and _writeBlock()
- update readme.md.
closed 2024-01-16 by RobTillaart