RobTillaart/FRAM_I2C #53: Different

board707, 2024-02-23:
Hello
As I wrote on the forum, I interesting in your code, because you take into account different types of addresses for chips of different sizes. Oddly enough, most similar libraries do not have this.

But why make separate classes for Fram9, Fram11 and Fram16? And where then are the classes for Fram8 and Fram10? :)
In fact, the principle of addressing i2c FRAM/EEPROM chips is the same for any size and follows a simple rule:
FM24с02 - has 8 bits cell address
FM24c04 - 9 bits
c08 - 10 bits
c16 - 11 bit
c32 -12 bit
c64 -13 bit
...
c512 - 16 bit
c1024 - 17bit
c2048 - 18 bit

For chips c02-c16, one byte is used for the cell address, and additional bits are added to the device i2c address.
For chips from c32 to c512 - two bytes are used.
For c1024 and c2048 chips - additional bits, in addition to two bytes, are added to the device i2c address, as is done for c04-c16 chips

Thus, it is quite possible to make a single library for all chip sizes. To use the correct addressing for each size, it seems to me that the simplest thing is to ask the user to add a chip size parameter to the initialisation of the FRAM object.

RobTillaart, 2024-02-24:
The main reason the library is as it is is that it followed an evolutionary path as most of my libraries do.
They often start with a (personal) need for a specific device and often only a subset of the functionality in a sketch.
Writing a library has been a good method for me to learn to know a device, and it abstracts the device.
The initial need gives rise to the initial release of the library, in which I look beyond my needs to improve the interface.

For FRAM (16 bit) the first version was early 2018 and only in 2022 FRAM32 was added on request. 
I recall that the footprint would be larger if the FRAM32 would become the only class.

_small test to see the difference now.
Compiling testFRAM.ino with FRAM for UNO ==> 7380 bytes, 
Compiling testFRAM.ino with FRAM32 for UNO ==> 7864 bytes.  
484 bytes difference so substantial. Don't recall if the diff was same order of magnitude then._

So I decided FRAM32 would be a derived class, the extra footprint would only affect those that needed the 32 bit .
It would not break the interface, keep things backwards compatible, that's good.

FRAM9 and FRAM11 were added last year (2023) upon request. 
As FRAM32 was a derived class, FRAM9 and FRAM11 become too.
In hindsight these two might have been one class with minimal overhead (address mask == size -1), 
so yes you should know the size to make it easy.

> And where then are the classes for Fram8 and Fram10? :)

There has not been a request for those yet, so maybe in the future. 
Not read the data sheet, it might be that they are covered by the FRAM9 and FRAM11.

So, it is very well possible to make one class to rule them all, however overhead and addressing schemes made me decide otherwise.

Hopes that this answers the questions a bit.

board707, 2024-02-24:
Thanks for the answer. The reasons why the library is built this way are quite clear.
However, if you are concerned about code efficiency and compactness, it might be worth considering eliminating the code duplication that I see in the library.
For example, the methods of write8, write16, write32 ... etc are repeated in the FRAM32 class, although they are already defined in the base FRAM class:
```
void FRAM::write8(uint16_t memAddr, uint8_t value)
{
  uint8_t val = value;
  _writeBlock(memAddr, (uint8_t *)&val, sizeof(uint8_t));
}

void FRAM32::write8(uint32_t memAddr, uint8_t value)
{
  uint8_t val = value;
  _writeBlock(memAddr, (uint8_t *)&val, sizeof(uint8_t));
}
```
 The whole difference between the FRAM and FRAM32 classes is only in the implementation of the ` _writeBlock` and `_readBlock` methods - so why duplicate the rest? In order for each class to call its ` _writeBlock`  method, declare the ` _writeBlock` and `_readBlock`  virtual.

A more advanced way of improvement is not to create separate methods for bytes, integers, long and float, but to write a single template method that accepts data of any type, as is done in the classic Arduino EEPROM library - see the put and get methods.

RobTillaart, 2024-02-24:
Agree, there is definitely room for improvement, and that is true for many of my libraries, 
A (serious) refactor including testing would take quite some time, time I rather spend on new libraries.

I also get requests to port libraries to ESP-idf, ESPhome, micropython et al, and just do not (choose to) have time.

RobTillaart, 2024-02-25:
> The whole difference between the FRAM and FRAM32 classes is ... ```writeBlock()``` ...

And  parameter ```uint32_t memAddr``` is 16/32 bit in several functions.

board707, 2024-02-25:
Correct, but you can use uint32 in both cases

closed 2025-06-21 by board707