RobTillaart/FRAM_I2C #54: FRAM size error

cynicalb, 2024-03-06:
Rob

This FRAM is unsupported, I'm just reporting a result here. I have a MB85RC64A on a custom board.  It's 8,192 word s × 8 bits FRAM.  When I run sizeInBytes = FRAM.getSize() * 1024; it returns 4096 bytes, half the capacity.  I will use the set function.

RobTillaart, 2024-03-06:
Thanks for reporting,
I will keep the issue open and when time permits dive into the datasheet to check where/why .

RobTillaart, 2024-03-06:
datasheet - https://www.fujitsu.com/us/Images/MB85RC64A-DS501-00019-2v0-E.pdf
does not mention a productID / device ID.

which label is on the device?
(from datasheet)
[image]

cynicalb, 2024-03-06:
My apologies, I see that I was working on the MB85RC64TA datasheet not the MB85RC64A which should be on my board.  You're right, the A version has no product or device ID.  

The plot thickens for me, I specified the MB85RC64A, but I have these markings, which is neither and I cannot find online. 
RC64A
E12214
   338

RobTillaart, 2024-03-07:
Did some search and could not find the meaning either.

Most of the time these codes exist of a product code, a production date and batch number. 
RC64A is clearly the product code.
12214 could be a date, however the example in the datasheet 11000 does not look like it
338 and 300 could be batch codes. 

Some codes from the datasheets I have gave a little info
RC16 E11050 300
RC128A E11000 300
C64TA 11900 700

RC64V E11150 300  ---> different types can have same code.
RC256V E11150 300

RC256V E11150 300  ---> first digit after the E is the reel size for ordering. (1500, 2000)
RC256V E21200 300

RC1MT E11300 300

And then Yes more info!

RC04: Product name
E11900: E1(Environment code) + 1900(Year and Week code)
300: 3(Factory code) + 00(Trace code)

[image]

Will add a line in the readme.md

cynicalb, 2024-03-07:
Thanks again for going above and beyond.  I've followed your work for some time.  In fact, if I see that you've written a library, I'll use that in preference over all others because your work is very good.  Plus you've done a few blogs that I think are defining works for the topic.  Plus you're a gentleman on forums and so on.  A big thanks for your contributions to my work!

I'm comfortable to say that the MB85RC64A declared with the FRAM32 class has been tested working with:
void write8(uint16_t memAddr, uint8_t value) uint8_t
void write16(uint16_t memAddr, uint16_t value) uint16_t
void write32(uint16_t memAddr, uint32_t value) uint32_t

uint8_t read8(uint16_t memAddr)
uint16_t read16(uint16_t memAddr)
uint32_t read32(uint16_t memAddr

tested with struct and array
uint16_t writeObject(uint16_t memAddr, T &obj) 
uint16_t readObject(uint16_t memAddr, T &obj)

Jason

closed 2024-03-07 by cynicalb

RobTillaart, 2024-03-08:
>  A big thanks for your contributions to my work!

You're welcome

cynicalb, 2024-03-09:
Rob

I wanted to send a gift but the payment failed.  Do you have a paypal
address that I can send it to?

Regards

Jason.

RobTillaart, 2024-03-10:
Thanks for you generous donation!