RobTillaart/FRAM_I2C #55: Fram address question

cynicalb, 2024-03-06:
Rob

I am trying to work out your addressing.  I have a MB85RC64A FRAM 64KBIT.  I note that this is unsupported on your list.  I can report that your library's basic functionality works on this chip other than getsize and getID.  I am trying to work out how your addressing works.  I ran this code below that writes a byte FF and then read FF and looks for a failure to write/read as the top of memory. 
    // Write Fram FF
   ```
 for (uint32_t addr = 0; addr < 1000000; addr++)
    {
        FRAM.write8(addr, 0xFF);
        uint8_t framtestread = FRAM.read8(addr);
        Serial.print(addr);
        Serial.print(":");
        Serial.println(framtestread);

        if (framtestread != 0xFF ) {
            Serial.println("fram read fail");
            break;
        }
    }
```  
**Results:**
0:255
...
131071:255
131072:0
fram read fail

I found that the FRAM.write and FRAM.read printed results up to 131071 Dec or 0x1FFF.  I can't work out how this relates to the 8,192 word s × 8 bits structure of the IC.

On a 1 byte read, why am I getting a result above the addr = 8196?

Thanks
Jason

RobTillaart, 2024-03-06:
> I can report that your library's basic functionality works on this chip 
good to hear

>  other than getsize and getID.

Think this can be seen as a known fact as the datasheet of the A and the V version are very similar.
from the readme.md
**uint16_t getSize()** returns the size in kiloBYTE. If the FRAM has no device ID register, the size cannot be read.

RobTillaart, 2024-03-06:
> I found that the FRAM.write and FRAM.read printed results up to 131071 Dec or 0x1FFF. I can't work out how this relates to the 8,192 word s × 8 bits structure of the IC.

Think the chip just masks or ignores the 3 upper bits of the address.
This makes address 0x01FFF the effective the same as  0x03FF , 0x07FF, 0x0FFF etc

datasheet
[image]

> On a 1 byte read, why am I getting a result above the addr = 8196?

same address wrapping.

cynicalb, 2024-03-07:
Okay thanks. Thanks for answering so quickly.

RobTillaart, 2024-03-07:
If this answers your question, you can close the issue.

closed 2024-03-07 by cynicalb