RobTillaart/FRAM_I2C #60: FRAM::clear seems to fail sporadically

GWiking, 2025-06-06:
Hi!

I have successfully used your library with 9-bit addressing on FRAM9.

I'm using it to store important parameters that need to be recalled after a power cycle.

To ensure that no residual data remains after the restore, I run FRAM::clear and immediately check that the addresses I'm using are set to 0.

However, occasionally—after many, many restarts—it seems to fail, and the data is not 0.

From what I can tell, the clear function is essentially a block write, and I haven't encountered any issues when performing individual read/write operations on specific addresses.

I was wondering if you have any insight into what might be happening, and whether there is a possible workaround.

Now I have introduced a retry of the clear and read-out.

Thank you!

RobTillaart, 2025-06-06:
Thanks for your question about clear().
At the moment I have no answer, as I never encountered it with my fram tests.

 hypothesis: timing

When do you call clear?
Right after reboot 
Or just before shutdown?

The fram might need a few extra millis when started.  

Have you looked into the datasheet if there are timing restrictions when started?
These might even be voltage dependent.

GWiking, 2025-06-07:
Hello Rob,

Thank you for your quick response.

There is a lot happening in the software before initialization and before reading/writing to the FRAM—several seconds spent setting up communication with a server via Ethernet, among other things. Right before the clear operation, I read out the stored parameters and log them, and they appear correct, indicating that the FRAM is functioning properly.

After reading, I perform a clear operation, as we don’t want any residual data remaining when recalling the parameters. As far as I can tell, I haven’t encountered issues with writing or reading individual addresses. I’m using an ESP32-S3, and since I’m only utilizing 20 positions in the FRAM, my current approach is to explicitly write those addresses to 0, verify that they are indeed 0, and if not, repeat the process.

I currently have 60 units deployed, each undergoing multiple restarts per day. This issue has occurred three times in the past month, making it difficult to consistently reproduce. I have bench tested it with stress test and no issues.

If the FRAM is not cleared correctly im throwing the unit out of the system because it is rather "fatal" to run with old data.

RobTillaart, 2025-06-07:
Thanks for the details,
You have 60 devices, running 30 days, several restarts, 20 bytes that is 10.000 clears and 200.000 byte writes.

Only three fails in so much writes imply the code of the library is probably OK. It is also simple code, not a lot could go wrong, and tested quite a lot.

Some questions to investigate deeper.
the three fails, 
- is that 1 byte or multiple bytes per fail?
- which of the 20?
- is it on same device? 2 or 3 devices?
- wires are all soldered?
- is there only a fail with ```clear()``` ?

(Don't know is a valid answer)

I2c...
Writing 20 zero's after one another implies a relatively long time LOW on the data line.
What pull up resistor do you use ?
are there signals near the I2C lines that may affect them?

Finally, would the library need ```bool clearAndVerify()```?

GWiking, 2025-06-07:
Thank you for the suggestions.

Actually Im mainly using the write32 and read32 as the unit uses the data as U32 but still 20 bytes of data. And after the clear Im using read8 for the 20 bytes.

Is that 1 byte or multiple bytes per fail? **-- > Not known**
which of the 20? **--> Not known**
is it on same device? 2 or 3 devices? **--> 1 device 2 times 1 at one occasion**
wires are all soldered? --> **we will try to replace one of them and investigate the HW, the SCL and SDA should be perfectly fine otherwise there will be a lot more problems.**
is there only a fail with clear() ? --> **yes as far as I have seen, just before clear() fails there is a succesfull read32.**

3k3 as pullup, 3.3V supply
bool clearAndVerify() could be a thing if it returns false in the occasion that even the read fails, as far as I know the function readX returns 0 if not read ok (tried with wrong address) and that is tricky to see the difference from a "normal 0"

RobTillaart, 2025-06-07:
> as far as I know the function readX returns 0 if not read ok (tried with wrong address) and that is tricky to see the difference from a "normal 0"

A solution is to add / implement some error handling e.g. 
- add  int _error;
- check I2C e.g. ```_error = _wire->endTransmissions()```
- add function ```int getLastError()```

Would allow at least some mechanism to detect errors.

A second step would break the interface - especially the read functions
- bool writeX(addr, value)
- bool readX(addr, &value)

GWiking, 2025-06-07:
Thank You for the suggestions.

I will in this case do 0 write of the first 4 x U32 and the 5th is an CRC32 of the 4 first that I will set to 0xFFFFFFFF as default.

I will check the data before going further and also redo if it fails.

I think this will cover my needs for now.

RobTillaart, 2025-06-11:
> I think this will cover my needs for now.

Feel free to close the issue

closed 2025-06-17 by GWiking

RobTillaart, 2025-06-17:
@GWiking 

Created/updated the develop branch( for 0.8.3) with additional error handling for the low level I2C functions.
Definitely not perfect but could give some insight especially when I2C fails.
Also implements additional returns after a failing I2C call either endTransmission() or requestFrom().

One function: to call  **int lastError()**

If you have time, please give it a try